boschsensortec/BNO055_SensorAPI #2: Integration with Micropython

visualecho1, 2017-10-11:
I've got one of these hooked up to an ESP32 and I'll be trying to get this driver working in Micropython.  Any insight by someone who's done something similar would be appreciated.

BoschSensortec, 2017-10-12:
Hi @visualecho1 ,

Which interface would you be controlling the sensor via? Also, any specific breakout boards that you will be utilizing?

visualecho1, 2017-10-12:
Oh, Hello!  I'm not sure I actually expected a response.  Thank you!  

I'm using Loboris' fork of Micropython ( here at https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo ) for Espressif ESP32-WROVER, but currently putting it on a ESP32-WROOM (smaller memory).  I have general purpose I/O lines connected to COM1, COM2, INT, and RESET, and otherwise configured for I2C operation.  The ESP32 can set any of those lines for SPI, I2C, or bit-banging, and I have Python code to call 'C' functions in the ESP32 I2C hardware driver.  I think I can cause an interrupt on the INT line, but not quite sure how to do that yet.

Using available examples, I need to synthesize the module driver that connects this sensor driver in 'C' to the MicroPython hardware I2C functions and the MicroPython import library space.   I don't know the Bosch BNO055 IMU sensor chip that well (YET!) and I'm pretty new to this fork of MicroPython, so I have a few things to learn and a few hundred lines of detailed code to write, however none of it looks particularly difficult to me... just lots of research and debugging.  Board is a new circuit I melted in a toaster last weekend, on single-layer aluminum.
[image]  
Okay sorry, I like showing off the shiny.

Here are key examples (these are for my reference as well)
I2C interfacing - this contains both maps to Python functions and the I2C calls themselves
https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/esp32/machine_hw_i2c.c
Bosch Sensortec BNO055 driver code (and this may not be the best place to put this)
https://github.com/MDVE/MicroPython_ESP32_psRAM_LoBo/tree/master/MicroPython_BUILD/components/micropython/extmod/bno055 
Module glue file (no commit yet)
MicroPython_BUILD/components/micropython/esp32/modbno055.c
an example of what to put in there is in the I2C driver as well as the MQTT module at
https://github.com/MDVE/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/esp32/modmqtt.c

visualecho1, 2017-10-12:
The IMU is taking I2C commands, so I know that's working.  By directly talking to it via Python I2C commands, I can set it to suspended power mode and read that register back correctly, here is a scope of the write:
[image]

BoschSensortec, 2017-10-13:
Hi @visualecho1 ,

That's a cool board! If you are familiar with the Arduino environment, you can find [here](https://github.com/arduino-org/brackets-arduino/tree/master/libraries/NAxesMotion/utility) an example of how to quickly configure the sensor. The NAxesMotion library is a wrapper based on the BNO055 code you have already used. 
The tricky part would be mapping the read-write functions. 
Additionally, you may not need to integrate the bno055_support.c file.

You will observe clock stretches on the bus. This can be reduced with a firmware update. Watch the [BNO055](https://www.bosch-sensortec.com/bst/products/all_products/bno055) page for updates.