boschsensortec/BNO055_SensorAPI #6: 'if' clause does not guard

davidljung, 2017-12-01:
Hello.  I attempted to compile the bno055.c file using GCC and discovered a number of possible bugs.
The compiler warns: 
`BNO055_driver/bno055.c:17181: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  if (prev_opmode_u8 != BNO055_OPERATION_MODE_CONFIG)
  ^~`

and, indeed, looking a the code:

	if (accel_unit_u8 != BNO055_ACCEL_UNIT_MG)
		com_rslt += bno055_set_accel_unit(BNO055_ACCEL_UNIT_MG);
		if (com_rslt == BNO055_SUCCESS) {
			/* Read the accel raw x data*/
			com_rslt += bno055_read_accel_x(&reg_accel_x_s16);
			if (com_rslt == BNO055_SUCCESS) {
				/* Convert the raw accel x to m/s2*/
				data_f =
				(float)(reg_accel_x_s16/BNO055_ACCEL_DIV_MG);
				*accel_x_f = data_f;
			} else {
			com_rslt = BNO055_ERROR;
			}
		} else {
		com_rslt = BNO055_ERROR;
		}
	return com_rslt;

It is indented as if the author thought that if accel_unit_u8 is == BNO055_ACCEL_UNIT_MG, then the following code wouldn't be executed.  However, since there is no { on the first line, only the second line depends on the if condition.  Either this is unintended behaviour or the author has uses misleading indentation as the compiler warns.

This warning is repeated hundreds of times as the same pattern repeats throughout the file.

BoschSensortec, 2017-12-04:
Hi @davidljung,

We are aware of quality issues related to older sensor drivers and are working to fix it. As mentioned by you, the author didn't intend it that way.

Blwnkl, 2018-11-04:
I'm trying to use the bno055.c code as well, and just stumbled upon this post.  Has this issue been corrected with new code?  (the code I have is version 2.0.3, 2016/03/14.
The last thing I want to do is debug a manufacturer's driver code! This should be tested and stable by this point in time.  Please advise.