boschsensortec/BNO055_SensorAPI #7: Memory requirements for the controller

rcacurs, 2017-12-15:
Hello! Could you provide the typical memory requirements of the host controller for this library (compile size, and requirements for RAM)? I think this is important information and is necessary when selecting host controller of embedded system integrating BNO055 sensor.

BoschSensortec, 2017-12-15:
Hi @rcacurs,

Do you have a specific processor architecture, compiler and compiler options in mind? It also depends on which APIs you use. For a given processor architecture, compiler and compiler options, we can only calculate the maximum memory requirements for the functions that are called and not the code in the application surrounding it.

rcacurs, 2017-12-15:
The plan is to use Texas Instruments  msp430 microcontrollers. For development we are using Code Composer Studio IDE and for compilation and will use the compiler tool-chain that comes with this IDE. I understand that exact FLASH memory requirement depends on compiler used, and it would be impossible to provide such information for all possible processor and compiler combinations, but I think it would be helpful if such information could be provided even for one processor architecture and compiler.    It would give at least approximate idea for memory requirements.

BoschSensortec, 2017-12-15:
Hi @rcacurs,

Assuming you are using all the functions (note the significant number of functions with overlapping functionality), for a cortex m0+ MCU with an optimization option -O0 on the arm-gcc compiler,
```
   text	   data	    bss	    dec	    hex	filename
  73736	      0	      4	  73740	  1200c	bno055.o (ex cortexm0plus\libapi.a)
   2780	      0	     24	   2804	    af4	bno055_support.o (ex cortexm0plus\libapi.a)
  76516	      0	     28	  76544	  12b00	(TOTALS)
 ```
and with an optimization option -O2,
```
   text	   data	    bss	    dec	    hex	filename
  32672	      0	      4	  32676	   7fa4	bno055.o (ex cortexm0plus\libapi.a)
    780	      0	     24	    804	    324	bno055_support.o (ex cortexm0plus\libapi.a)
  33452	      0	     28	  33480	   82c8	(TOTALS)
 ```
Note that the support file helps give an idea of memory requirements for the application code.

closed 2017-12-15 by rcacurs