boschsensortec/BNO055_SensorAPI #11: Is the code compatible with a Arduino pro Mini

DavidRTucker, 2018-08-03:
I am getting a warning from the compiler 

_' library BNO055 claims to run on (samd) architecture(s) and may be incompatible with your current board which runs on (avr) architecture(s).'_

Does this mean that the library will not work with my Arduino Pro Mini?

ma-harvey, 2018-08-03:
The code found here is generic C code and can be ported to any microcontroller platform. Arduino or not.

Since you are using Arduino, I recommend that you leverage the work of others before you and download a BNO055 Arduino library directly.

dhruvkakadiya, 2018-08-03:
BNO055 drivers are stand alone support for any kind of microcontrollers. But you'll need to make sure, you implement below functions for your Arduino Pro Mini board.

```
s8 BNO055_I2C_bus_write(u8 dev_addr, u8 reg_addr, u8 *reg_data, u8 cnt);
s8 BNO055_I2C_bus_read(u8 dev_addr, u8 reg_addr, u8 *reg_data, u8 cnt);
void BNO055_delay_msek(u32 msek);
```

DavidRTucker, 2018-08-04:
Thank you for your quick response.  You have given me hope!

 I appreciate that the driver would work with any micro controller however the .ino examples appear to be specific to Arduino with the setup and loop sections and it is code that I have simply copied into my sketch that is generating the compiling error.  
Does the patch code need to be added in the setup section of the example the Arduino sketches?

dhruvkakadiya, 2018-08-06:
Have a look at https://github.com/BoschSensortec/BNO055_driver/pull/8/commits/4a4ebde439838b07580f68beef20386c21501a27 commit which is patch for the compilation issue you are seeing.

HariMurti12, 2018-12-22:
I got a problem with the GY-955 sensor, I tried to use the code given directly by * Kriswiner Update BNO_055_Nano_Basic_AHRS_t3.ino * but what I use is Arduino Mega2560, I have replaced the pin to 18 / tx 19 / rx and I switch to 20 / it's 21 / scl but still doesn't respond, where is this error located,

or maybe I am wrong in the library i2c_t3.h because the error that appears is in that direction

please help with this problem, thank you