boschsensortec/BNO055_SensorAPI #12: BNO055 hardware fusion is wrong at low angular speed

schadocalex, 2018-10-29:
Hi, I use a BNO055 and I have a strange problem. At low angular speed, the intern kalman does not change the yaw value. If I turn manually the imu at about 3deg/s, the angle and the gyroscopes values returned by the fusion are static (there is some very few noise near 0). But above 5-10 deg/s it's working as expecting (and very well).

If I read raw input (without fusion), gyroscope values become good even at low speed.

Important note: I don't use magnetometer because of the huge metallic mass near the imu (the BNO055 is in IMU mode for fusion (acc+gyro)).

ma-harvey, 2018-10-29:
There is a threshold to prevent drift under small movements. Since the noise is not perfectly Gaussian, the integration when the signal is close to zero would drift very fast.

The built-in sensor fusion software of the BNO055 does not handle such use-case. Usually I would recommend eCompass mode, but since this is also not an option, thenI would recommend to look into other types of sensor fusion algorithms, and program them into the part with BMF055.

BNO055 can also be reprogrammed, after removing the security bit which erases the firmware.

schadocalex, 2018-10-29:
Thanks for the quick asnwer.

I understand the underlying problem of fusing small noised values. If I undestrand well, the threshold you talk about is on accelerometers values, isn't it? Because gyroscope yaw value, for the angle fusion, is good, but yes the imu is near 0 in acc x,y,z.

I will try to fuse data in software to see if its better than the built-in kalman without magnetometer!

Thanks.

closed 2018-10-29 by schadocalex

ma-harvey, 2018-10-30:
The no-motion interrupt is  _hardware feature_ of the BMA280 accelerometer inside of the BNO055. The threshold is a _software feature_ of the BSX3 sensor fusion library that runs on the SAMD20 MCU inside the BNO055.

This threshold is not configurable, and changing it would conflict with the gyroscope automatic offset calibration routine, which also uses the accelerometer to detect a standing still motion, in order to measure the gyroscope offset.

If the offset calibration was done manually, it would be possible do lower that threshold like what you did in your test, while still achieving good results, but to do both at the same time is quite a challenge, and operation over the entire temperature range would most likely affect performance more than normal.

The bottom-line is : for BNO055 the configuration is fixed. The thresholds cannot be changed, and it is suited to a wide range of applications in a wide range of environments. Offset compensation routines are also always enabled.